crjens/PiPowerMeter #1: Current Sensors, Config Elements, and Sensor Coverage.

akkord64, 2017-07-12:
I have the dl-ct08cl5 CT's (2000/1) and am unable to configure the readings properly. I also dont seem to understand the scaling factors or where the phase configurations are defined (V1,V2, V3, V1+V2). The CT's seem to be detect voltage and frequency, but no current. I am measuring ~150ohm across the CT's leads before connecting to anything. I am measuring ~44 vac and ~7.6M ohm on a 1AMP load.

Also - For any 220V circuits, do we need both legs? Just curious how the off the shelf clamp meters are calculating load when using only a single leg?

craigjensen, 2017-07-12:
The CS5463 chip outputs voltage and current readings as a number in the range [0-1] which is then multiplied by the voltage and current scale factors to arrive at the actual voltage and current readings.  The scale factors depend on various characteristics of the measurement circuits and the turns ratio of the transformer.  I haven’t used the dl-ct08cl5 CT’s you reference but since they have a 2000/1 turns ratio (same as the YHDC sct-013 100A), you should start with a scale factor of 100 and fine tune as needed.

When you configure your circuits you’ll need to make sure you select the correct voltage reference (V1 or V2) or else you won’t get accurate results.  Take a look at the waveform chart and you’ll be able to see pretty quickly if the voltage and current waveforms are in phase.  If not, just change the voltage reference and check it again.  Also, make sure the CT is installed correctly or else you’ll get negative results.

There are two ways to measure 240V circuits.  If the circuit doesn’t have a neutral the current on each leg must be equal so you can use a single CT on one of the legs and set the voltage reference to V1+V2.  If the circuit has a neutral you’ll get more accurate results by using a CT on each leg with the voltage reference set to V1 for one leg and V2 for the other.

akkord64, 2017-07-19:
Thanks for the reply. Let me try a few things based on your response and post an update. Also I have the power unit currently powered by 120v instead of 220v which *may* be influencing the readings? I will change that and continue with my tuning.

craigjensen, 2017-07-19:
The power supply is designed to work with either 120V or 240V so that shouldn’t influence the results

akkord64, 2018-02-10:
Well after having my 4th child I am back to trying to get this project working again. I seem to have hit a wall on this as I cannot get any sensible readings from my unit. 

I am not testing with a SCT-019 and was trying to measure a single lef of mains. I have a burden resistor on the probe @26 Ohms (calculated @200amps here: http://tyler.anairo.com/projects/open-energy-monitor-calculator) and had to scale the voltage (366) and current (26) to get a waveform that looks like this (what is the current wave doing??)

[image]

After a couple of days of frustration with trying to get this to make sense of this (as well as moving between ports, burden resistors, additional boards, Pi's) I decided to unplug all sensors and I am seeing readings (when running server.js directly) of

pi@PowerPi:~/app$ node server.js 
next rollup in: 19056531
spawned reader with pid: 2755
2018-02-10T15:42:23.558Z loading configuration...
2018-02-10T15:42:23.560Z waiting for table: Circuits
2018-02-10T15:42:23.561Z waiting for table: Config
2018-02-10T15:42:23.568Z opened database: powermeter.db
2018-02-10T15:42:23.593Z currentSha: c67445f87196163bc68588797bb8e7412fd9d357
2018-02-10T15:42:23.594Z currentDate: 2017-12-14T04:19:55.000Z
2018-02-10T15:42:23.614Z elapsed 0.042 (PRAGMA foreign_keys=on)
2018-02-10T15:42:23.614Z elapsed 0.043 (create table if not exists Probes ( id INTEGER primary key, Type text, Board int check(Board>=0 and Board<=7), CurrentChannel int check(CurrentChannel>=0 and CurrentChannel<=15), VoltageChannel int check(VoltageChannel>=0 and VoltageChannel<=3), Breaker int, Alert Text, SourceType int);)
2018-02-10T15:42:23.617Z elapsed 0.045 (create table if not exists Config ( Name text primary key, Value Text);)
2018-02-10T15:42:23.618Z Config table ready
2018-02-10T15:42:23.619Z elapsed 0.048 (create table if not exists Circuits ( id INTEGER primary key, Name Text, Description Text, Enabled int, IsMain int, Probes text);)
2018-02-10T15:42:23.620Z Circuits table ready
2018-02-10T15:42:23.623Z elapsed 0.007 (pragma table_info(Probes);)
2018-02-10T15:42:23.625Z Probes table ready
2018-02-10T15:42:23.625Z elapsed 0.008 (Insert or ignore into Config Values('Port', '3000');)
2018-02-10T15:42:23.626Z elapsed 0.008 (Insert or ignore into Config Values('DeviceName', '');)
2018-02-10T15:42:23.626Z elapsed 0.008 (Insert or ignore into Config Values('Price', '0.1');)
2018-02-10T15:42:23.626Z elapsed 0.008 (Insert or ignore into Config Values('Region', 'en-US');)
2018-02-10T15:42:23.626Z elapsed 0.008 (Insert or ignore into Config Values('VoltageScale', '384');)
2018-02-10T15:42:23.626Z elapsed 0.008 (Insert or ignore into Config Values('Probes', '[{''Name'':''SCT-006'',''Factor'':''40''},{''Name'':''SCT-010'',''Factor'':''152''},{''Name'':''SCT-013'',''Factor'':''101''},{''Name'':''SCT-016'',''Factor'':''152''},{''Name'':''SCT-019'',''Factor'':''295''}]');)
2018-02-10T15:42:23.627Z elapsed 0.008 (Select * from Config where Name = 'BackupDB';)
2018-02-10T15:42:23.627Z elapsed 0.007 (Insert into Circuits (id, Name, Description, Enabled, IsMain, Probes) select null,'Circuit 1','',1,0,'1' where (select count(*) from Circuits) = 0;)
2018-02-10T15:42:23.628Z elapsed 0.007 (create table if not exists Readings (id INTEGER primary key, CircuitId int, I real, V real, P real, Q real, PF real, Timestamp int, Compacted int, F real, foreign key(CircuitId) references Circuits(id));)
2018-02-10T15:42:23.629Z elapsed 0.011 (Insert or ignore into Config Values('Mode', '000060');)
2018-02-10T15:42:23.629Z elapsed 0.011 (Insert or ignore into Config Values('Config', '001001');)
2018-02-10T15:42:23.631Z elapsed 0.007 (Insert into Probes (id, Type, Board, CurrentChannel, VoltageChannel, Breaker, Alert, SourceType) select 1,'SCT-006',0,0,0,20,null,0  where (select count(*) from Probes) = 0;)
2018-02-10T15:42:23.632Z elapsed 0.004 (pragma table_info(Readings);)
2018-02-10T15:42:23.633Z Readings table ready
2018-02-10T15:42:23.634Z elapsed 0.001 (select count(*) as Rows from Readings;)
2018-02-10T15:42:23.635Z elapsed 0.002 (create index if not exists Readings_Timestamp_CircuitId_P_idx on Readings(Timestamp, CircuitId, P);)
2018-02-10T15:42:23.636Z elapsed 0.003 (create index if not exists Readings_CircuitId_Timestamp_P_idx on Readings(CircuitId, Timestamp, P);)
2018-02-10T15:42:23.666Z elapsed 0.002 (select * from Circuits;)
2018-02-10T15:42:23.668Z elapsed 0.003 (Select * from Config;)
2018-02-10T15:42:23.669Z username:     password: 
2018-02-10T15:42:23.670Z Setting port: 3000
2018-02-10T15:42:23.691Z App Server running at port 3000
2018-02-10T15:42:23.691Z elapsed 0.024 (Select * from Probes;)
2018-02-10T15:42:23.695Z elapsed 0 (Select * from Config;)
2018-02-10T15:42:23.696Z read: DeviceName->
2018-02-10T15:42:23.696Z read: Mode->000060
2018-02-10T15:42:23.697Z read: Region->en-US
2018-02-10T15:42:23.697Z read: VoltageScale->384
2018-02-10T15:42:23.697Z read: Port->3000
2018-02-10T15:42:23.697Z read: Price->0.1
2018-02-10T15:42:23.697Z read: Config->001001
2018-02-10T15:42:23.701Z mqtt: undefined
reader closed 1
reader closed 2
Opening device: /dev/spidev0.0 at 2000000 Hz, mode 0, 8 bits, 0 us delay
Device opened with mode: 0, bits: 8, speed: 2000000
Device opened: Hardware version: 2.2
RESET
Register dump:
0 - Config: 001001
1 - CurrentDCOffset: 000000
2 - CurrentGain: 400000
3 - VoltageDCOffset: 000000
4 - VoltageGain: 400000
5 - CycleCount: 000fa0
[... 254 lines cut ...]
2018-02-10T15:43:38.749Z sending cached config
[... cut ...]
2018-02-10T15:43:48.356Z  : V= 130.5  I= 0.0  P= -1.6  Q= 2.7  PF= -0.4256  F= 60.000  F2= 60.000 (500 samples in 124.724679ms)
2018-02-10T15:43:51.139Z Circuit 1 : V= 130.5  I= 0.0  P= -1.5  Q= 2.6  PF= -0.4276  F= 59.999  F2= 60.000 (500 samples in 124.721122ms)
^C2018-02-10T15:43:52.256Z Received kill signal, shutting down gracefully.

This seems to make NO SENSE with a voltage reading and NO probes inserted!! Do you think I have faulty hardware? I also tested the dev branch and dont see a difference on the empty probe readings.

No Probe waveform:
[image]

craigjensen, 2018-02-10:
Hi Chris,

I have tested the SCT-019 sensors and they work well but the PiPowerMeter boards have built-in sense resistors so you should not be adding your own.  I’ve seen some pretty unusual current waveforms especially if you have lots of electronics connected that might account for what you are seeing. Please try removing any additional burden resistors and connect the current probes directly to the current board.  That should give you a more sensible reading.

The voltage waveform is read directly from the connected power source.  The current sensors only read current, not voltage, which is why you still see a voltage waveform when not current sensors are connected.

Let me know if you are still having issues and I’ll help however I can.

Craig

[bad-wave]<https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F10292093%2F36063875-86a7d77a-0e50-11e8-9564-a2e56c8ba673.JPG&data=04%7C01%7Ccjensen%40microsoft.com%7Cf2e932c93e7340a27b1708d570a06c01%7Cee3303d7fb734b0c8589bcd847f1c277%7C1%7C0%7C636538756662671218%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwifQ%3D%3D%7C-1&sdata=srDdY8l4sOr9H96mPHO6RKrF%2F9NbqhnPbLu2psWgh5M%3D&reserved=0>

[noprobe-wave]<https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F10292093%2F36063995-68bca1ee-0e52-11e8-9204-3ab8b9cb4af7.JPG&data=04%7C01%7Ccjensen%40microsoft.com%7Cf2e932c93e7340a27b1708d570a06c01%7Cee3303d7fb734b0c8589bcd847f1c277%7C1%7C0%7C636538756662671218%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwifQ%3D%3D%7C-1&sdata=LBdHNhsD3Do0DIUjzQcFq5AzlCN9GZLpmXe2dJw%2FQBc%3D&reserved=0>

akkord64, 2018-02-11:
I finally have success! I changed the unit to power from 220v (the documentation is not clear that the voltage of each leg is derived from the unit INPUT power. I connected AC1,AC2,Neutral and I can now see Phase A and B. After removing burdens, I also see current and wattage values which are approximate. I now understand your scaling logic and how it is applied in both relation to voltage and current. I am at a point of major relief! I think a few updates to the documentation should be made to call out the importance of the following:

1: the units AC Source is how voltage is derived for each phase.
2: Physical Wiring (no need for burdens)
3. How the CS5463 readings are taken (high level sequencing, but explain the scale multipliers. I initially thought "scale" referred to peak values of readings and not as a multiplier)

I also have some things that are not clear yet.

1. How does Type: Source, Load, Both factor into the reporting/values.
2. Phase - What exactly does this do? Is it used in calculations or just as a reference?
3. VARS - I have some values which are negative. I am not sure if this means my CT's are backwards? I get conflicting data between the type value (#1) and watts/vars.

Suggestions:
1. Breaker configs - Would like to see breakers as a configurable element (like you did for probes)
2. Pinned header with navigational controls - Navigating back and forth between config/waveform/circuits.
3. MQTT configurable

As a side note - I had to scale my SCT-019's to 615 to get an approx accurate reading. I currently only have 1, but have ordered more and will report on the accuracy of the other units when they arrive. Your seeded scale value of 295 is nowhere near 615 and I wouldn't think that much variability would happen. Either way I will report back my progress in hopes it helps someone else. before closing the isssue.555555555 My current waveforms do not look sinusodal; I assume this is because the loads are different. Some being motors, others being AC/DC switching supplies, etc.

I am also sending you a request for a new current board as I am running out of ports (I thought I could just measure one leg of my 220v circuits, but I now see the need for measuring both).

crjens, 2018-02-12:
Glad you were able to get it working and thanks for the great feedback, I’ll definitely try to improve the documentation in the future.

To answer your specific concerns:

  1.  SourceType is currently only used to help clean up noise on the readings when there is no load on the circuit (i.e. loads should never have negative values).  This may change in the future but for now that’s all it’s used for.
  2.  Phase is important because that’s where you specify which voltage reference to use for the power calculations.  If you are seeing strange readings such as a load circuit that appears to be producing energy (negative watts), it’s probably because either; 1) your CT is backwards, or 2) you are using the wrong voltage reference.
  3.  VAR’s stands for Volt-Amp reactive<https://en.wikipedia.org/wiki/Volt-ampere_reactive> and is a measure of how much reactive current is flowing in the circuit.  Inductive loads such as electric motors typically have a current waveform that lags the voltage producing a negative power factor<https://en.wikipedia.org/wiki/Power_factor>.  Capacitive loads do the opposite so depending on what kind of load you have connected you could see either negative or positive vars flowing on a particular circuit.  Connecting the CT’s backwards will also affect this value so you should make sure all CT’s connected correctly.  In the case of YHDC sensors make sure the side of the sensor that has writing on it faces toward the source.
  4.  You shouldn’t need to double the scale factor in the SCT-019 to get an accurate reading.  If you only have one sensor and want to estimate the usage across both legs of your panel, I’d suggest leaving the scale factor for the sensor alone and setting the phase to V1+V2.  This will give you a decent estimate as long as the usage is divided relatively equally across the two legs.  Using two sensors, one on V1 and the other on V2, is the best way to get an accurate reading if the current is not divided equally.
  5.  The meter does have rudimentary MQTT support.  Just add ?advanced=1 to the config page url you you’ll be able to set the MQTT server address (http://yourserver:3000/configure.html?advanced=1)

Hope this helps clear things up a bit.

Regards,
Craig