crjens/PiPowerMeter #2: Potential issue with your current sense resistors

High5Apps, 2017-07-25:
Hey Craig,

First off, thanks for open sourcing this project! I've recently been using it as a reference to develop a power monitor with 6 CS5463s for 3 phases of solar generation and 3 phases of household consumption.

Unfortunately, I think that I may have found a potential issue when examining your hardware specs. In your [PowerMeterCurrent_DigikeyBOM.txt](https://github.com/crjens/PiPowerMeter/blob/master/Eagle/V2.2/PowerMeterCurrent_DigikeyBOM.txt) it indicates that the value of your current sensing resistors is 5 ohms. In your documentation you recommend using [sct-013-000 100A:50mA](https://raw.githubusercontent.com/crjens/PiPowerMeter/master/Documentation/DSC_0002.JPG) current transducers. Furthermore, on page 7 of the [CS5463 datasheet](http://www.mouser.com/ds/2/76/cs5463_f3-48303.pdf), it indicates that the differential current inputs can handle up to a 500mVpp difference (+/- 250mV since the signals are ground tied).

From those numbers I'm getting that the CS5463 would only be able to measure up to a maximum of 70.8A of sinusoidal input, not 100A before the CS5463 current readings start "peaking". I believe that you may have forgotten to divide by sqrt(2) when choosing your 5 ohm current sense resistor. My reasoning for dividing by sqrt(2) when going from a peak-peak to RMS can be found at the bottom of the [Average Electrical Power section of the Root Mean Square](https://en.wikipedia.org/wiki/Root_mean_square#Average_electrical_power) Wikipedia page where it states that Vrms = Vp / sqrt(2).

If the CS5463 can measure a max of 250mV peak, then it can only handle a 250mVp / sqrt(2) = 177mVrms sinusoidal signal without "peaking". Dividing this by your current sense resistor of 5 ohms gives 177mVrms / 5 ohms = 35.4 mArms through your current sense resistor. Finally, multiplying that by the turns ratio for the sct-013-000 of 100A:50mA (N = 2000) gives 35.4mArms * 2000 = 70.8Arms through the current transducer before "peaking" begins to occur. 

I believe that if you were to substitute a 5.00 ohm / sqrt(2) = 3.5 ohm resistor, you would again be able to read the full 100Arms range of the sct-013-000.

I am quite new to hardware of this complexity, so it is likely that I have made a mistake in dividing by sqrt(2), but I wanted to bring this to your attention nonetheless. Have you been able to take measurements above 70.8 A of sinusoidal current? While currents this high are unlikely to be seen under normal circumstances, I believe that the PiPowerMeter would not be able to handle them. 

Thanks again for your time and for developing this amazing project,
Julian

craigjensen, 2017-07-26:
Your calculations look correct for the sense resistor alone but they don't account for the influence of other components such as the multiplexers and filters in the circuit.  During prototyping I experimented with different values of the sense resistor until arrived at a value that closely matched the full scale reading of the sct-013 with that of the cs5463 and it turned out to be around 5 ohms.  If you look at the software scale factor used for the sct-013 sensor you'll see that it defaults to 101 which means that when the cs5463 outputs a full scale current reading (Irms = 1.0) the resulting current reading will be 101A.  That said, 3.5 ohms would have worked just as well and provided a little more headroom to measure higher current levels.  I'm sorry I don't have a more scientific analysis on how I arrived at the value of 5 ohms but sometimes a little trial and error works as well as anything :)

I'm glad you found my project informative and I'd love to see more details on what you are building if you are willing to share.

High5Apps, 2017-07-27:
Oh you're right- I definitely forgot to include the other resistances in my calculations. Since I have 6 CS5463s for 6 inputs, I don't have any multiplexers on that path, but I'll definitely need to account for the current sense filters. 

As far as the project goes, I've been working on developing a python/Raspberry Pi-based energy monitor on-and-off for the past year or so. Previously, I had just been using an ADC to sample 4 sct-013-030 current sensors (3 phases of consumption 1 phase of generation), and then multiplying the values by the "expected" voltage (230V here). But a few months ago I decided I wanted to start taking voltage into account. I also wanted the option to go up to 3 phases of generation. During my research I found your project and learned about CS5463s for the first time. My requirements differ from yours in 2 main ways. First, I only need to measure 6 inputs, not 128 (although I think the stackable boards are a really neat idea). Second, I didn't want to do the round-robin sampling, but since I'm only doing 6 inputs and the CS5463s are relatively cheap, I just incorporated 6 of them. I'm planning to develop a machine learning energy disaggregation system to remove the need for more than 6 inputs. 

My first prototype board (with 5 ohm current sense resistors) is supposed to arrive tomorrow, so I'll finally be able to experimentally test out the scales. If you don't mind, do you remember what your experimental setup was for finding the voltage and current scales? I imagine it would be something like this:
- Power the Raspberry Pi using the micro USB cable instead of through the PSU
- Attach a function generator to the screw terminals, producing a known current and voltage at 50 (or 60) Hz
- Place the sct-013-000 around the positive cable of the function generator
- Read at least 1 computational cycle of RMS voltage and current through the CS5463, 
- Divide the expected rms voltage or current by the measured values on the scale [-1, 1)

Am I missing anything? Thanks again for your help.

craigjensen, 2017-07-27:
That's pretty much the way I did it - attach known voltage and current and adjust scale factors to match.  I'm not sure if you are interested in reading the instantaneous V/I values but one thing to look out for is that the Raspberry Pi may have problems reading those values from 6 CS5463's at the same time.  It may work but that's a lot of interrupts to process...  If you limit to only reading the values at the end of the computation cycle you should have no issues.

If you haven't already, you might want to take a look at the CS5464 and CS5467 IC's.  They are able to measure multiple channels and may reduce the number of chips you need to use.

High5Apps, 2017-07-27:
Great thanks, I just wanted to make sure I wasn't off base with that setup. 

I opted away from interrupts in my design because I was worried about that as well. To reduce complexity, I've just muxed the 6 chip selects and I'm planning to poll the status registers to see when a new conversion cycle measurement is ready. 

I looked into the CS5463 and 67 when I first learned about the Cirrus website, but if I remember correctly, I don't think that those 2 chips really add extra inputs. It seemed like they just added redundancy to the single-phase measurement for tamper-protection. While tamper protection isn't a priority for me, I did briefly look into the ADE7878, which someone mentioned on one of your forum posts. Unfortunately, its complexity was a little too intense for my EE skills, but I believe it truly does take 3 phases of input as well as performing tamper protection. 

One further question- did you have to do any phase compensation? If so what was your setup for experimentally determining it? Again, thanks a lot!

craigjensen, 2017-07-27:
I have experimented with phase compensation to account for the non-linearity in the CT's.  If you look [here](https://learn.openenergymonitor.org/electricity-monitoring/ct-sensors/yhdc-ct-sensor-report) you'll see a graph that indicates the SCT-013 has a phase shift of ~4 degrees for a 22ohm shunt resistor.  Another approach I took was to hook up a purely resistive load and then adjust until the reported power factor was as close to 1.0 as I could get.  Neither of these is likely going to be 100% accurate and if you need super accurate results you'll probably want to look into using [Rogowski coils](https://en.wikipedia.org/wiki/Rogowski_coil) instead of CT's to measure current.

High5Apps, 2017-08-17:
Sorry for the delayed response, I haven't had as much time to work on this recently as I'd hoped.

I have finally been able to experimentally test the voltage and current scales for my prototype board. 
My concerns about the 3.5 Ohm vs 5 Ohm burden resistors were unfounded, and so I am closing this issue.

Using known RMS currents and voltages, I determined that the current scales for my 6 inputs were between 102A and 105A, which should allow full scale current measurements over the SCT-013-000's linear range. My voltage scales were between 357V and 360V, which also allows for full scale voltage measurements. These results are consistent with your scales (101A, 384V), given that my design was somewhat modified from yours.

I was also able to do phase compensation for the SCT-013-000s using large resistive loads, achieving power factors of between 0.99990 and 0.99999.

I've still got more work to do, but I'm eventually hoping to open source my CS5463 python library too.

Thanks again for your help and for sharing your project with all of us!

closed 2017-08-17 by High5Apps