crjens/PiPowerMeter #8: Are these board purchaseable someplace?

xekon, 2019-07-28:
Are these board purchaseable someplace?

This looks to be the best raspberry pi based power monitor I could find on the net.

the only other one I could find was this: https://store.ncd.io/product/12-channel-on-board-95-accuracy-ac-current-monitor-with-i2c-interface/

but this project looks more flexible and even has a way to monitor the mains in the breaker panel.

crjens, 2019-07-29:
I have boards available.  Please DM me at crjens@hotmail.com<mailto:crjens@hotmail.com> for details

presslab-us, 2020-03-17:
Are these still available?  I'm not sure if the email listed is still valid.  I sent an email 5 days ago but have yet to get a reply.

crjens, 2020-03-17:
Yes – I do have boards available.  Sorry if I missed your previous  mail.  It may have gone into my junk folder and I missed it.  Please email again at crjens @ hotmail.com<mailto:crjens@hotmail.com> and I’ll send details.

rogerfar, 2021-04-12:
@crjens do you still have some boards?

crjens, 2021-04-13:
Hi Roger,

I do have boards available for purchase.  The control board is $75 and each 16 channel input board is $25.  Shipping is $10 in the US or $20 outside the US.  The boards come fully assembled and tested.  You'll need to supply your own Raspberry Pi and current sensors.

If you are interested in purchasing, please contact me directly at ***@***.******@***.***> and I'll send payment details.

Thanks,
Craig

fxstein, 2021-07-13:
Sent you an email. Thanks!