crjens/PiPowerMeter #11: MQTT Provision

xianrecords2015, 2019-12-04:
Hello,

Thanks for making this project. I started my own with a standard MCP3208 but found that a 12bit ADC is not accurate enough. I like the approach of the CS5490.

I have a question for you. In the driver.js, I saw that you have some provision for MQTT but I could not find anywhere on the configuration web page a way to enter the MQTT server, username, password and port. Am I missing something ?

Thanks

Chris

crjens, 2019-12-04:
It’s not exposed by default.    Add ‘?advanced=1’ to the url and you should see it

xianrecords2015, 2019-12-04:
Thank you for your answer,

I was trying to analyze the CS5490.js and forgive my ignorance and little knowledge of Javascript but you are requiring a CS5490 at the beginning of the code. I am assuming that this is not the CS5490.js file itself. There are a number of call to a cs5490 library (ie: cs5490.DigitalRead) that I cannot reference. Where are they coming from ?

Also you have define 4 input pins to read the HW version. I have a hard time to find them on the schematics. I can find the output pins but I am not understanding the inputs.

Thanks

Christian

xianrecords2015, 2019-12-04:
I now understand the 4 input pins for HW version.

I also see that you have developed a CS5490 C++ in one of the other branch that I believe is what you require in the cs5490.js. I am wondering in the setup file how you import it in the project and the setup

crjens, 2019-12-05:
The 4 input pins are to determine the hardware version of the boards.  There are drivers written in C for both the CS5463 and CS5490 chips.  They use nan<https://github.com/nodejs/nan> to interface with the node based PiPowerMeter software.

xianrecords2015, 2019-12-11:
There seems to be some inconsistencies in the V4 Schematics/Board.

- Are the shunt resistance R1-R16 5.1Ohms or 7.5Ohms ? They are listed 5 Ohms in the schematics yet they call a 7.5Ohms Digikey Part number in the info

- the PJ313D have the CP1-3525N-ND Digikey Part number in V2 BOM (There is no V4 BOM). These seems to be thruholes connectors. But the board is setup for surface mount connector. Do you have the reference of these SMT connectors ?

Tx