crjens/PiPowerMeter #18: Software installation errors

Jendemen, 2020-12-29:
Hi,
Just before I wanted to order the components I try to install the software, but whatever I try, I keep getting errors as below.

Does anyone has suggestions? I am running 2020-02-13-raspbian-buster-lite

Many thanks in advance, Jan

make: *** [CS5490.target.mk:109: Release/obj.target/CS5490/CS5490.o] Error 1
make: Leaving directory '/home/pi/app/node_modules/CS5490/build'
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/home/pi/.nvm/versions/node/v14.15.3/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (events.js:315:20)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:277:12)
gyp ERR! System Linux 4.19.97-v7+
gyp ERR! command "/home/pi/.nvm/versions/node/v14.15.3/bin/node" "/home/pi/.nvm/versions/node/v14.15.3/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/pi/app/node_modules/CS5490
gyp ERR! node -v v14.15.3
gyp ERR! node-gyp -v v5.1.0
gyp ERR! not ok
>>> Enable UART
>>> Disable bluetooth
Removed /etc/systemd/system/multi-user.target.wants/hciuart.service.
>>> CS5490 is installed.  Please reboot to apply changes.

> sqlite3@5.0.0 install /home/pi/app/node_modules/sqlite3
> node-pre-gyp install --fallback-to-build

node-pre-gyp WARN Using request for node-pre-gyp https download
node-pre-gyp WARN Tried to download(403): https://mapbox-node-binary.s3.amazonaws.com/sqlite3/v5.0.0/napi-v3-linux-arm.tar.gz
node-pre-gyp WARN Pre-built binaries not found for sqlite3@5.0.0 and node@14.15.3 (node-v83 ABI, glibc) (falling back to source compile with node-gyp)
make: Entering directory '/home/pi/app/node_modules/sqlite3/build'

elektrolubach, 2020-12-29:
Hi @Jendemen,

I recently had the same errors during installing the PiPowerMeter software.
The last error is causing the above mentioned errors I think.
I did not deep dive in, but this must have to do with the type of API for NodeJS (and its constraints and requirements) which is used here.

In the driver C++-Modules for the CS5463 (separate repository [here](https://github.com/crjens/cs5463), file CS5463.cc, line 517) or the CS5490 ([here](https://github.com/crjens/CS5490), file CS5490.cc, line 110) I changed the following line:
`info.GetReturnValue().Set(String::NewFromUtf8( isolate, version.c_str() ));`
to
`info.GetReturnValue().Set(String::NewFromUtf8( isolate, version.c_str() ).ToLocalChecked());`
I changed these files locally after downloading the software to my Pi using the wget command.

Afterwards I re-built these modules the PiPowerMeter software with `npm install` and I got it working.

Maybe this helps you as well...
Cheers elektrolubach

Jendemen, 2020-12-30:
Hi @elektrolubach,

That did the job. Albeit lots of warnings I now can start the webpage.

Thanks a lot!

closed 2021-01-02 by Jendemen