crjens/PiPowerMeter #22: Fresh install errors / no readings

chillr21, 2022-09-03:
Hi there,

Just tried to do a fresh install of the current version of PiPowerMeter but can not get it to work correctly (I'm back to using my old version via a disk img I had created back in 2019 - `PiPowerMeter Software timestamp: 018-11-26T02:58:19.000Z`).

The installer seems to complete without any errors, it does hang for a while here and there thou.

Once installed and rebooted the web page is accessible but no power readings can be obtained and the nodejs.err.log has the following error:

```
Error: Cannot find module '/home/pi/app/node_modules/CS5490/build/Release/CS5490.node'. Please verify that the package.json has a valid "main" entry
    at tryPackage (node:internal/modules/cjs/loader:364:19)
    at Function.Module._findPath (node:internal/modules/cjs/loader:577:18)
    at Function.Module._resolveFilename (node:internal/modules/cjs/loader:942:27)
    at Function.Module._load (node:internal/modules/cjs/loader:804:27)
    at Module.require (node:internal/modules/cjs/loader:1028:19)
    at require (node:internal/modules/cjs/helpers:102:18)
    at Object.<anonymous> (/home/pi/app/CS5490.js:3:14)
    at Module._compile (node:internal/modules/cjs/loader:1126:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1180:10)
    at Module.load (node:internal/modules/cjs/loader:1004:32) {
  code: 'MODULE_NOT_FOUND',
  path: '/home/pi/app/node_modules/CS5490/package.json',
  requestPath: 'CS5490'
}
node:events:491
      throw er; // Unhandled 'error' event
      ^

Error [ERR_IPC_CHANNEL_CLOSED]: Channel closed
    at new NodeError (node:internal/errors:387:5)
    at ChildProcess.target.send (node:internal/child_process:742:16)
    at /home/pi/app/driver.js:170:20
    at /home/pi/app/database.js:874:37
    at Statement.<anonymous> (/home/pi/app/database.js:605:17)
Emitted 'error' event on ChildProcess instance at:
    at node:internal/child_process:746:35
    at processTicksAndRejections (node:internal/process/task_queues:78:11) {
  code: 'ERR_IPC_CHANNEL_CLOSED'
}
node:internal/modules/cjs/loader:372
      throw err;
      ^

Error: Cannot find module '/home/pi/app/node_modules/CS5490/build/Release/CS5490.node'. Please verify that the package.json has a valid "main" entry
    at tryPackage (node:internal/modules/cjs/loader:364:19)
    at Function.Module._findPath (node:internal/modules/cjs/loader:577:18)
    at Function.Module._resolveFilename (node:internal/modules/cjs/loader:942:27)
    at Function.Module._load (node:internal/modules/cjs/loader:804:27)
    at Module.require (node:internal/modules/cjs/loader:1028:19)
    at require (node:internal/modules/cjs/helpers:102:18)
    at Object.<anonymous> (/home/pi/app/CS5490.js:3:14)
    at Module._compile (node:internal/modules/cjs/loader:1126:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1180:10)
    at Module.load (node:internal/modules/cjs/loader:1004:32) {
  code: 'MODULE_NOT_FOUND',
  path: '/home/pi/app/node_modules/CS5490/package.json',
  requestPath: 'CS5490'
}
```

Fresh install details:
Rpi 3B 1GB
SD card flashed with official image - "2021-03-04-raspios-buster-armhf-lite.img"
Logged on as pi then executed: `wget -O - https://raw.githubusercontent.com/crjens/PiPowerMeter/master/setup.sh | bash`

soup-campbell, 2023-09-06:
Same Problem. Error: Cannot find module '/home/pi/app/node_modules/CS5490/build/Release/CS5490.node  Power Meter does not start.. No entries made to powermeter.db. In addition, some of the dependancies should be updated.
wiringpi- is no longer current

overpaul, 2024-12-23:
Chillr21 - any hance of getting a copy of your SD backup image?  I lost my SD card and can not reload the system.  I have my data, but no SD card image.

chillr21, 2024-12-23:
> Chillr21 - any hance of getting a copy of your SD backup image? I lost my SD card and can not reload the system. I have my data, but no SD card image.

Sure, how do you propose I share it for you to grab it?

overpaul, 2024-12-24:
Thats great.

Can you upload it to Dropbox or Google Drive or another file sharing option.
If you give me your email, I can open up a folder on my Google Drive that
you can upload to
I could also mail you an SD card that you can copy the image to.

However we do it, it will save me a lot of work and save my power monitor.
Thanks

*Paul Oversmith*

***@***.*** ***@***.***>*
*941-266-3600*

chillr21, 2024-12-24:
OK, I've tried sending an SMS with details to your phone number

soup-campbell, 2024-12-24:
It will take me a couple of days to locate it. I have not looked at it in over a year.-Soup-

chillr21, 2024-12-24:
Or send the link to this address if it doesn't get filtered out:
pigekey120
at
chosenz
dot
com

soup-campbell, 2024-12-24:
I am sorry, I don't think I have what you were looking for. My interest was writing a datalogger centered around the CS5490. The human interface and server portions of that posting were of no interest to me. I never did pursue the problems I was having with the download.-Soup-

 
 

chosenx

stampeder, 2025-03-23:
Has anyone got this working recently?
Thanks.
Glenn

soup-campbell, 2025-03-23:
no

overpaul, 2025-03-23:
> no <br> On Sunday, March 23, 2025 at 01:12:41 AM GMT-5, Glenn

I had a bad SD card and had to reload the software. I couldn't get it to
work. I posted on the forum, and a nice person sent me an image, which I
loaded and got it to work. I can't remember what the problem was; it could
have been outdated support software. I did save an image of the restored
version so that if I had a failure, I could go back to it.

*Paul Oversmith*

***@***.*** ***@***.***>*
*941-266-3600*

stampeder, 2025-03-23:
Could you send me a copy of the image?
Thanks.

overpaul, 2025-03-23:
Pleased to the file is 4 gig in size. (winRAR).
Not sure I can upload it on GitHub.
It has some custom mods to meet my needs, but it is basically the same.

I can try with email if you send me an e-mail address.

*Paul Oversmith*

***@***.*** ***@***.***>*
*941-266-3600*

stampeder, 2025-03-24:
Thanks Paul. My email is gtracy@peoplescom.net. Let's try that.