crjens/PiPowerMeter #25: Does someone still have some boards available

pleuba, 2023-12-23:
I'm using the power meter for several years, and really liked it, but recently faced an issue on an jack connector.

I'm looking more specifically for the 16 channel input board. I tried to contact @crjens by email, but without success.

So if someone has one available, please contact me.

Philippe

jburdy, 2023-12-23:
I can't help you, but I wanted to share that I'm worried. Like you, I'm very happy with it, it's been running for 5 years. 
I tried to contact crjens last year to buy another one as a backup but got no response...

I wonder if we shouldn't kindly find a better distributed alternative...

stampeder, 2025-05-05:
I know this is an old thread. I have had my pipowermeter controller board and one i/o board sitting for several years and  I would like to get them going. I need a working image of an sd card that I can spin up please. I have tried installing from scratch without success. 
Thanks.
Glenn...

pleuba, 2025-09-22:
> [@pleuba](https://github.com/pleuba) [@jburdy](https://github.com/jburdy) guys i've got a powerboard and 16ch available if you still want one.

Yes, I’m intrrested

stampeder, 2025-09-22:
gtracy@peoplescom.net

pleuba, 2025-09-22:
> [@pleuba](https://github.com/pleuba) Mate whats your email? @rhyst9 pleuba_at_swissonline.ch